luigifcruz/pisdr-image #1: Extra hardware support

italic-r, 2019-07-29:
I saw in your list of supported software you include GQRX and GNUradio, and in supported hardware I see a few older Airspy devices. SDRplay and the AirspyHF+ devices are pretty popular, and they are supported by GQRX and GNUradio through the osmocom module. [willcode/gr-osmosdr](https://github.com/willcode/gr-osmosdr/tree/sdrplay2) includes SDRplay support and Alexandru Csete has accepted AirspyHF+ support in [csete/gr-osmosdr](https://github.com/csete/gr-osmosdr-gqrx/tree/airspyhf). I got these two trees merged pretty easily for use in GQRX. 

From the license terms in the SDRplay drivers, you are able to distribute the binary as long as you include the license. Airspy looks to follow something similar.

yo3hjv, 2019-12-19:
Hello Luigi and thanks for this nice project!
Unfortunately, I tried to use V3.0  on a RaspberryPi 4B/4 GB but at no avail...
The error I get at the start is: 
[end Kernel Panic - not syncing: No working int found. Try passing init= option to kernel. See Linux Documentation/admin-guide/init.rst for guidance. ]

Then it stops.
I used FAT32 on a 16Gb SD card, also tried with a 32 Gb SDHC. Tried exFAT also. Same result. The SD card are Smasung EVO type.

Do you have any advice on this?

Best 73 de Adrian YO3HJV

luigifreitas, 2019-12-19:
> Hello Luigi and thanks for this nice project!
> Unfortunately, I tried to use V3.0 on a RaspberryPi 4B/4 GB but at no avail...
> The error I get at the start is:
> [end Kernel Panic - not syncing: No working int found. Try passing init= option to kernel. See Linux Documentation/admin-guide/init.rst for guidance. ]
> 
> Then it stops.
> I used FAT32 on a 16Gb SD card, also tried with a 32 Gb SDHC. Tried exFAT also. Same result. The SD card are Smasung EVO type.
> 
> Do you have any advice on this?
> 
> Best 73 de Adrian YO3HJV
> 
> LE: Same problem on a Raspi 3B+.

Hello Adrian, 

Sorry to hear about your problem. I couldn't replicate this error. Have you tried installing the image using [Balena Etcher](https://www.balena.io/etcher/)?

73's Luigi

luigifreitas, 2020-02-20:
The Airspy HF+ is now supported.

closed 2020-02-20 by luigifreitas