luigifcruz/pisdr-image #6: pisdr v3 on RPI2B+

9a4db, 2019-12-21:
Hello,
Here is copy from terminal on SDRangel start / GNU Radio is able to start but can not detect LimeSDR USB, Lime suite also not able to start.... 
=>
pi@pisdr:~/Packages/sdrangel/build $ sudo ./sdrangel
libEGL warning: DRI2: failed to authenticate
2019-12-21 16:18:49.949 (W) QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
2019-12-21 16:18:49.964 (I) SDRangel 4.12.2 Qt 5.11.3 32b arm Raspbian GNU/Linux 10 (buster) DSP Rx:24b Tx:16b PID 2498
2019-12-21 16:18:49.967 (I) MainSettings::MainSettings: settings file: format: 0 location: /root/.config/f4exb/SDRangel.conf
2019-12-21 16:18:49.968 (D) AudioDeviceManager::AudioDeviceManager: scan input devices
2019-12-21 16:18:50.404 (D) AudioDeviceManager::AudioDeviceManager: input device #0: default
2019-12-21 16:18:50.404 (D) AudioDeviceManager::AudioDeviceManager: input device #1: jack
2019-12-21 16:18:50.405 (D) AudioDeviceManager::AudioDeviceManager: input device #2: pulse
2019-12-21 16:18:50.405 (D) AudioDeviceManager::AudioDeviceManager: input device #3: usbstream:CARD=ALSA
2019-12-21 16:18:50.405 (D) AudioDeviceManager::AudioDeviceManager: input device #4: alsa_output.platform-soc_audio.analog-mono.monitor
2019-12-21 16:18:50.405 (D) AudioDeviceManager::AudioDeviceManager: scan output devices
2019-12-21 16:18:51.712 (D) AudioDeviceManager::AudioDeviceManager: output device #0: default
2019-12-21 16:18:51.712 (D) AudioDeviceManager::AudioDeviceManager: output device #1: jack
2019-12-21 16:18:51.713 (D) AudioDeviceManager::AudioDeviceManager: output device #2: pulse
2019-12-21 16:18:51.713 (D) AudioDeviceManager::AudioDeviceManager: output device #3: sysdefault:CARD=ALSA
2019-12-21 16:18:51.713 (D) AudioDeviceManager::AudioDeviceManager: output device #4: dmix:CARD=ALSA,DEV=0
2019-12-21 16:18:51.713 (D) AudioDeviceManager::AudioDeviceManager: output device #5: dmix:CARD=ALSA,DEV=1
2019-12-21 16:18:51.714 (D) AudioDeviceManager::AudioDeviceManager: output device #6: dmix:CARD=ALSA,DEV=2
2019-12-21 16:18:51.714 (D) AudioDeviceManager::AudioDeviceManager: output device #7: dsnoop:CARD=ALSA,DEV=0
2019-12-21 16:18:51.714 (D) AudioDeviceManager::AudioDeviceManager: output device #8: dsnoop:CARD=ALSA,DEV=1
2019-12-21 16:18:51.715 (D) AudioDeviceManager::AudioDeviceManager: output device #9: dsnoop:CARD=ALSA,DEV=2
2019-12-21 16:18:51.715 (D) AudioDeviceManager::AudioDeviceManager: output device #10: hw:CARD=ALSA,DEV=0
2019-12-21 16:18:51.715 (D) AudioDeviceManager::AudioDeviceManager: output device #11: hw:CARD=ALSA,DEV=1
2019-12-21 16:18:51.715 (D) AudioDeviceManager::AudioDeviceManager: output device #12: hw:CARD=ALSA,DEV=2
2019-12-21 16:18:51.716 (D) AudioDeviceManager::AudioDeviceManager: output device #13: plughw:CARD=ALSA,DEV=0
2019-12-21 16:18:51.716 (D) AudioDeviceManager::AudioDeviceManager: output device #14: plughw:CARD=ALSA,DEV=1
2019-12-21 16:18:51.716 (D) AudioDeviceManager::AudioDeviceManager: output device #15: plughw:CARD=ALSA,DEV=2
2019-12-21 16:18:51.716 (D) AudioDeviceManager::AudioDeviceManager: output device #16: usbstream:CARD=ALSA
2019-12-21 16:18:51.717 (D) AudioDeviceManager::AudioDeviceManager: output device #17: alsa_output.platform-soc_audio.analog-mono
2019-12-21 16:18:51.719 (D) MainWindow::MainWindow: start
2019-12-21 16:18:52.231 (D) MainWindow::MainWindow: load settings...
2019-12-21 16:18:52.232 (D) MainWindow::loadSettings
2019-12-21 16:18:52.233 (W) qUncompress: Input data is corrupted
2019-12-21 16:18:52.234 (W) qUncompress: Input data is corrupted
2019-12-21 16:18:52.234 (W) qUncompress: Input data is corrupted
2019-12-21 16:18:52.235 (D) AudioDeviceManager::deserialize
2019-12-21 16:18:52.235 (W) qUncompress: Input data is corrupted
2019-12-21 16:18:52.236 (D) AMBEEngine::deserialize: invalid or no data
2019-12-21 16:18:52.236 (W) qUncompress: Input data is corrupted
2019-12-21 16:18:52.240 (D) MainWindow::MainWindow: load plugins...
2019-12-21 16:18:52.241 (D) PluginManager::loadPluginsPart folder /home/pi/Packages/sdrangel/build/../lib/sdrangel/plugins is empty
2019-12-21 16:18:52.246 (D) PluginManager::loadPluginsDir: fileName: libchanalyzer.so
2019-12-21 16:18:52.257 (I) PluginManager::loadPluginsDir: loaded plugin libchanalyzer.so
2019-12-21 16:18:52.258 (D) PluginManager::loadPluginsDir: fileName: libdemodam.so
2019-12-21 16:18:52.270 (I) PluginManager::loadPluginsDir: loaded plugin libdemodam.so
2019-12-21 16:18:52.271 (D) PluginManager::loadPluginsDir: fileName: libdemodatv.so
2019-12-21 16:18:52.282 (I) PluginManager::loadPluginsDir: loaded plugin libdemodatv.so
2019-12-21 16:18:52.283 (D) PluginManager::loadPluginsDir: fileName: libdemodbfm.so
2019-12-21 16:18:52.302 (I) PluginManager::loadPluginsDir: loaded plugin libdemodbfm.so
2019-12-21 16:18:52.303 (D) PluginManager::loadPluginsDir: fileName: libdemoddatv.so
2019-12-21 16:18:52.878 (I) PluginManager::loadPluginsDir: loaded plugin libdemoddatv.so
2019-12-21 16:18:52.879 (D) PluginManager::loadPluginsDir: fileName: libdemoddsd.so
2019-12-21 16:18:52.895 (I) PluginManager::loadPluginsDir: loaded plugin libdemoddsd.so
2019-12-21 16:18:52.895 (D) PluginManager::loadPluginsDir: fileName: libdemodfreedv.so
2019-12-21 16:18:52.909 (I) PluginManager::loadPluginsDir: loaded plugin libdemodfreedv.so
2019-12-21 16:18:52.910 (D) PluginManager::loadPluginsDir: fileName: libdemodlora.so
2019-12-21 16:18:52.919 (I) PluginManager::loadPluginsDir: loaded plugin libdemodlora.so
2019-12-21 16:18:52.920 (D) PluginManager::loadPluginsDir: fileName: libdemodnfm.so
2019-12-21 16:18:52.933 (I) PluginManager::loadPluginsDir: loaded plugin libdemodnfm.so
2019-12-21 16:18:52.934 (D) PluginManager::loadPluginsDir: fileName: libdemodssb.so
2019-12-21 16:18:52.946 (I) PluginManager::loadPluginsDir: loaded plugin libdemodssb.so
2019-12-21 16:18:52.947 (D) PluginManager::loadPluginsDir: fileName: libdemodwfm.so
2019-12-21 16:18:52.958 (I) PluginManager::loadPluginsDir: loaded plugin libdemodwfm.so
2019-12-21 16:18:52.959 (D) PluginManager::loadPluginsDir: fileName: libfilesource.so
2019-12-21 16:18:52.973 (I) PluginManager::loadPluginsDir: loaded plugin libfilesource.so
2019-12-21 16:18:52.973 (D) PluginManager::loadPluginsDir: fileName: libfreqtracker.so
[... 99 lines cut ...]
2019-12-21 16:18:53.815 (D) PluginManager::registerTxChannel  Remote channel source  with channel name  "sdrangel.channeltx.remotesource"
2019-12-21 16:18:53.815 (D) PluginManager::registerSampleSource  Remote device input  with source name  sdrangel.samplesource.remoteinput
2019-12-21 16:18:53.816 (D) PluginManager::registerSampleSink  Remote device output  with sink name  sdrangel.samplesink.remoteoutput
2019-12-21 16:18:53.816 (D) PluginManager::registerRxChannel  SSB Demodulator  with channel name  "sdrangel.channel.ssbdemod"
2019-12-21 16:18:53.816 (D) PluginManager::registerTxChannel  SSB Modulator  with channel name  "sdrangel.channeltx.modssb"
2019-12-21 16:18:53.817 (D) PluginManager::registerSampleSource  SoapySDR Input  with source name  sdrangel.samplesource.soapysdrinput
2019-12-21 16:18:53.817 (D) PluginManager::registerSampleSink  SoapySDR Output  with sink name  sdrangel.samplesink.soapysdroutput
2019-12-21 16:18:53.818 (D) PluginManager::registerSampleMIMO  Test Multiple Input  with MIMO name  sdrangel.samplemimo.testmi
2019-12-21 16:18:53.818 (D) PluginManager::registerSampleSink  Test Sink Output  with sink name  sdrangel.samplesink.testsink
2019-12-21 16:18:53.819 (D) PluginManager::registerSampleSource  Test Source input  with source name  sdrangel.samplesource.testsource
2019-12-21 16:18:53.819 (D) PluginManager::registerRxChannel  UDP Channel Sink  with channel name  "sdrangel.channel.udpsink"
2019-12-21 16:18:53.820 (D) PluginManager::registerTxChannel  UDP Channel Source  with channel name  "sdrangel.channeltx.udpsource"
2019-12-21 16:18:53.820 (D) PluginManager::registerRxChannel  WFM Demodulator  with channel name  "sdrangel.channel.wfmdemod"
2019-12-21 16:18:53.821 (D) PluginManager::registerTxChannel  WFM Modulator  with channel name  "sdrangel.channeltx.modwfm"
2019-12-21 16:18:53.823 (D) DeviceEnumerator::enumerateRxDevices: sdrangel.samplesource.airspy
2019-12-21 16:18:53.860 (D) AirspyPlugin::enumOriginDevices: enumerated 0 Airspy devices AIRSPY_ERROR_NOT_FOUND
2019-12-21 16:18:53.861 (D) AirspyPlugin::enumOriginDevices: airspy_exit: AIRSPY_SUCCESS
2019-12-21 16:18:53.861 (D) DeviceEnumerator::enumerateRxDevices: sdrangel.samplesource.airspyhf
2019-12-21 16:18:53.900 (D) DeviceEnumerator::enumerateRxDevices: sdrangel.samplesource.fileinput
2019-12-21 16:18:53.901 (D) DeviceEnumerator::enumerateRxDevices: sdrangel.samplesource.fcdpro
2019-12-21 16:18:53.939 (D) DeviceEnumerator::enumerateRxDevices: sdrangel.samplesource.fcdproplus
2019-12-21 16:18:53.942 (D) DeviceEnumerator::enumerateRxDevices: sdrangel.samplesource.kiwisdrsource
2019-12-21 16:18:53.943 (D) DeviceEnumerator::enumerateRxDevices: sdrangel.samplesource.limesdr
Illegal instruction
pi@pisdr:~/Packages/sdrangel/build $

9a4db, 2019-12-21:
Also ->
pi@pisdr:~/LimeSDR/LimeSuite/build/LimeUtil $ sudo ./LimeUtil --info
######################################################
## LimeSuite information summary
######################################################

Version information:
  Library version:	v19.04.1-ga5b3a10f
  Build timestamp:	2019-12-15
  Interface version:	v2019.1.0
  Binary interface:	19.04-1

System resources:
  Installation root:	/usr/local
  User home directory:	/root
  App data directory:	/root/.local/share/LimeSuite
  Config directory:	/root/.limesuite
  Image search paths:
     - /root/.local/share/LimeSuite/images
     - /usr/local/share/LimeSuite/images

Supported connections:
Illegal instruction
pi@pisdr:~/LimeSDR/LimeSuite/build/LimeUtil $ 

Regards
Djani

snowfog, 2019-12-23:
It is also the same issue with gqrx with a rtl-sdr dongle. 

pi@pisdr:~ $ gqrx
libEGL warning: DRI2: failed to authenticate
gr-osmosdr 0.1.5 (0.1.5) gnuradio 3.7.13.4
built-in source types: file fcd rtl rtl_tcp rfspace airspy soapy redpitaya 
Resampling audio 96000 -> 48000
BookmarksFile is /home/pi/.config/gqrx/bookmarks.csv
Illegal instruction

9a4db, 2019-12-23:
Right...
I started from clean minimal Raspibian img
Now have Soapy and GNU radio up and running
for limeNET micro

luigifreitas, 2019-12-23:
Error caused by Cortex-A72 (RPi 4) instructions trying to run on a Cortex-A53 (RPi 3). This issue will be fixed in the next release. If you don't want to wait, recompiling LimeSuite and Soapy should fix the problem.

`
cd LimeSDR/LimeSuite
sudo rm -r build
mkdir build
cd build
cmake ..
make -j4
sudo make install
sudo ldconfig
`

closed 2019-12-23 by luigifreitas