luigifcruz/pisdr-image #9: BalenaEtcher does not recognize image

wjgilmore, 2019-12-28:
Hi,

Today I downloaded the PiSDR image from both the official and a collaborator mirror (US-East) via https://pisdr.luigifreitas.me/#getting-started, and in both cases when I attempt to flash the image onto a 16GB SD card using the macOS version of BalenaEtcher (v1.5.70) I receive this message:

```
Invalid image
/Users/wjgilmore/Downloads/pisdr_v3.0.tar.xz is not a supported image type.
```

I subsequently tried downloading the Torrent (my first Torrent ever lol, used Transmission client), and wound up with the same result. I've burned dozens of Raspberry Pi images using Balena Etcher and this is the first time I've encountered this issue. Any ideas?

Thank you!
Jason

wjgilmore, 2019-12-28:
The problem was due to macOS' default archive utility. I installed The Unarchiver (a free app in the app store), and it extracted the .img file no problem. Once extracted, balenaEtcher recognized the file.

closed 2019-12-28 by wjgilmore

luigifreitas, 2019-12-29:
Thanks for reporting this issue, I'm make sure this is corrected for the next release.