luigifcruz/pisdr-image #13: GNURadio is not working with PlutoSDR

heikipoika, 2020-01-23:
Im running PiSDR 3.0 on Pi4. SDRAngel works with my PlutoSDR, but not GNURadio. When I add the PlutoSDR Sink or Source Blocks to my flow graph and press execute I get the following error:

/*--------------------------*/
<<< Welcome to GNU Radio Companion 3.7.13.4 >>>

Block paths:
	/usr/share/gnuradio/grc/blocks
	/usr/local/share/gnuradio/grc/blocks

Loading: "/home/pi/Slask/transmitter.grc"
>>> Done

Generating: '/home/pi/Slask/top_block.py'

Executing: /usr/bin/python2 -u /home/pi/Slask/top_block.py

Traceback (most recent call last):
  File "/home/pi/Slask/top_block.py", line 26, in <module>
    from gnuradio import iio
ImportError: cannot import name iio

>>> Done
*/-----------------------------*/

citizenrich, 2020-01-26:
The 3.7.13.4 version of GNU Radio has had a great deal of work done on it since its release which was on 2018-07-15. So perhaps a way to isolate the issue is to use the maint-3.7 branch or latest master to see if it sorts things out. Just a suggestion.

Krounet, 2020-01-31:
Hello,

I had this problem too, and I found a solution here [https://ez.analog.com/wide-band-rf-transceivers/design-support/f/q-a/106867/can-t-get-pluto-gnuradio-file-to-work/316749](url)

To resume the python path for the iio package is not at the good location. To solve this, you can change its location by doing this :
`cp -r /usr/local/lib/python2.7/dist-packages/gnuradio/iio /usr/lib/python2.7/dist-packages/gnuradio/`

It works fine for me now.

luigifreitas, 2020-02-01:
Thanks for reporting this bug. It will be corrected in the next release. Until there, @Krounet's suggestion should make `gr-iio` work again.

luigifreitas, 2020-02-01:
This issue will be closed once fixed.

heikipoika, 2020-02-12:
@Krounet's fix works for me as well! Thankyou!

luigifreitas, 2020-02-20:
Fixed, it'll be available in the next release.

closed 2020-02-20 by luigifreitas