luigifcruz/pisdr-image #19: Gqrx blacklist for rtl_v3

kf4mot, 2020-03-11:
**Describe the bug**
Gqrx can't open rtl v3 dongle. Adding blacklist as described here fixes it.
[https://nw-digital-radio.groups.io/g/main/wiki/GQRX-SDR](url)

This really isn't so much a bug as a config issue. If a fix isn't appropriate then maybe some documentation in the hardware section would do.

**Device Version**
Pi4 2gb running 4beta2.

luigifreitas, 2020-03-15:
This is weird, as far as I know, I [configured](https://github.com/luigifreitas/pisdr-image/blob/a26e61c5f96cb22a7eed480822fd2776768c4b73/builder/stage4/09-pisdr-rtlsdr/00-run.sh#L14) the compiler to automatically detach the kernel drivers. Must be something local with GQRX. I'll do some testing...

kf4mot, 2020-03-15:
I should add that 2 devices were already blacklisted. I had to add my V3 as the third one.
I added blacklist dvb_usb_rtl28xxu

Dougga, 2020-05-18:
My bug, just filed, may be a duplicate

luigifcruz, 2020-08-15:
Closing since the next version will have updated drivers. Thanks for contributing!

closed 2020-08-15 by luigifcruz