luigifcruz/pisdr-image #21: Pls add support for Blade RF

Jeffrey-S, 2020-04-25:
Thank you for your work with this RPi image.  Would it be possible to include support for the Nuand BladeRF series of radios?

luigifreitas, 2020-05-26:
Thanks for trying out PiSDR! I'm planning to add BladeRF support as soon as possible. I appreciate your suggestion.

closed 2021-04-13 by luigifcruz

MountainLogic, 2021-04-13:
The Soapy interface provide an semi-universal set of drivers so that the problem only needs to be solved once.

luigifcruz, 2021-04-13:
@MountainLogic Yes, that's right. But I need to manually validate if the device actually works as intended on every release. At the moment, I can't do this because I don't have a BladeRF unit. Soapy is aleady installed on every PiSDR release.

MountainLogic, 2021-04-13:
@luigifcruz , Oh ,didn't realize you supported Soapy.  Way cool.