luigifcruz/pisdr-image #22: No audio using bluetooth speakers or headset

Dougga, 2020-05-18:
**Describe the bug**
Connection can be successfully made using the User Interfaces to bluetooth audio devices, but no sound can be made to go through any of them.

**To Reproduce**
Steps to reproduce the behavior:
1. Install image file on sd card
2. Update all packages
3. Enable Bluetooth
4. Make Bluetooth speaker bluetooth visible
5. Pair Raspberry Pi 4 with speaker
6. Run GQRX and  tune to a strong FM broadcast channel.

**Expected behavior**
Music should come out of the bluetooth device as it does on my HF rig

**Screenshots**
not terribly relevant - imagine a strong signal

**Device Version**
4b

**Additional context**
Urban setting with absurdly strong FM signals bouncing about

luigifreitas, 2020-05-26:
Thanks for reporting this bug! This might be a Raspbian fault, this image uses the same building process as the official release. Does the Bluetooth device work on other mainstream apps like Chromium?

luigifcruz, 2020-08-15:
Closing for lack of response. Debian is likely the cause of the bug. Next version will feature a revised audio library support.

closed 2020-08-15 by luigifcruz