luigifcruz/pisdr-image #32: PiSDR Version 4.0 Refresh won't boot at RPI4 - 4Gb

Laav, 2020-08-03:
**Describe the bug**
Flashed the "2020-08-02-PiSDR-vanilla.img.xz" image four times with Raspberry Pi Imager 1.4 (verified with the app), on an brand new SanDisk Ultra microSDHC UHS-1 Card - 32GB (https://www.action.com/nl-nl/p/sandisk-micro-sdhc-kaart-3/). Tried to boot an Raspberry PI 4 - 4Gb without luck.

After that i've extracted the .xz file to flash the "2020-08-02-PiSDR-vanilla.img" file another time (fifth), this time also without luck and no boot/network.

After five faillures i've tried the "2020-03-08-PiSDR-vanilla.img", this on booted right after the flash with success and was accessible with Putty.

**To Reproduce**
Flash "2020-08-02-PiSDR-vanilla.img.xz" or "2020-08-02-PiSDR-vanilla.img" with Raspberry Pi Imager 1.4 (on an SanDisk Ultra microSDHC UHS-1 Card - 32GB).

**Expected behavior**
Boot and use the new 2020-08-02 version.

**Device Version**
Raspberry PI 4 - 4Gb

Stealthbird97, 2020-08-07:
Might be worth a try using https://www.balena.io/etcher/ 
Just installed on my Pi4 8GB and it boots fine.

tomtomate68, 2020-08-11:
This is a old Version from GNU 3.7.14 Please make a new Version thx

luigifcruz, 2020-08-15:
@Laav Looks like your image got corrupted while downloading. Try downloading it again. I recommend doing it with [balenaEtcher](https://www.balena.io/etcher/).

Laav, 2020-08-15:
Downloaded the file again and flashed with balenaEtcher, the whole is booting fine now, thanks for that.

closed 2020-08-15 by Laav

luigifcruz, 2020-08-15:
You're welcome, thanks for trying PiSDR!