luigifcruz/pisdr-image #34: Document USRP setup

duggabe, 2020-08-10:
The following steps are required to set up a USRP device:
```
cd ~/
git clone git://github.com/EttusResearch/uhd.git
cd uhd
git checkout v3.15.0.0
cd ~/uhd/host/utils
sudo cp uhd-usrp.rules /etc/udev/rules.d/
cd /etc/udev/rules.d
sudo chown pi:pi *.rules
sudo udevadm control --reload-rules
sudo udevadm trigger
cd ~/uhd/host/utils
# plug in USRP
uhd_usrp_probe
```
Once that is done, it works well. Tested with 2020-08-05-PiSDR-vanilla.img and a USRP B200mini.

luigifcruz, 2020-08-14:
Thanks! The [latest build](https://github.com/luigifcruz/pisdr-image/actions/runs/207602504) has all these steps built-in now, including the `udev` rules. It should work out-of-the-box. Could you please test it with the actual hardware for me?

duggabe, 2020-08-16:
testing of PiSDR 2020-08-13-PiSDR-vanilla.img
-   the rules in /etc/udev/rules.d/ should be owned by pi rather than root
-   I still had to do 'sudo uhd_images_downloader'
-   and 'uhd_usrp_probe' which probably has to be done after the USRP is plugged in
-   see the complete process for building UHD at https://wiki.gnuradio.org/index.php/InstallingGRFromSource_on_Raspberry_Pi#Install_UHD_from_source

luigifcruz, 2020-08-16:
I thought `uhd_images_downloader` wasn't required. I'll make the required changes. Thanks again for testing!

duggabe, 2020-11-28:
I loaded and tested https://github.com/luigifcruz/pisdr-image/actions/runs/360794128
The USRP B200mini works out of the box now.
The Pluto works out of the box.
The only new problem is related to links to LOCALE being missing.

luigifcruz, 2020-12-02:
Thanks for testing! The LOCALE problem will be fixed.
Final USRP support will be released in PiSDR v5.0. Closing!

closed 2020-12-02 by luigifcruz