luigifcruz/pisdr-image #40: Ubuntu 20.10 based PiSDR for arm64 and x86 with headless (server) option

vielmetti, 2020-11-14:
**Is your feature request related to a problem? Please describe.**

@luigifcruz suggests this 

What do you guys think of a PiSDR image based on Ubuntu 20.10 with ARM64 and X86 flavors?

https://twitter.com/luigifcruz/status/1327486307791228928

**Describe the solution you'd like**

Put together a Ubuntu 20.10 based distribution of PiSDR, and target both arm64 and x86 devices.

**Describe alternatives you've considered**

Run with the current Ubuntu 20.04 based system on Pi 4.

**Additional context**

Seemed like a good idea when @luigifcruz  mentioned it on Twitter, so I thought I'd capture some of the discussion here too.

seuliang, 2021-01-10:
very support.
the raspbian (Raspberry Pi OS) is 32 bits, and lack of information to solve occured problems compared to ubuntu.
why not use ubuntu provided on offical website:
https://www.raspberrypi.org/software/operating-systems/

luigifcruz, 2021-04-13:
Development of this version has started and should be added as a branch pretty soon!

luigifcruz, 2021-12-31:
I'm scrapping the x86 version of PiSDR for now. But the ARM64 version is almost ready and should be released by next week if the v6.0 launch goes according to plan.

closed 2021-12-31 by luigifcruz