luigifcruz/pisdr-image #46: CygnusRFI doesn't work (Python2)

charlieh0tel, 2020-12-20:
Version: 2020-11-13-PiSDR-vanilla.img

CygnusRFI has some Python 2'isms in it.   But gnuradio, on which it depends, is set up for Python3.

Steps to reproduce the behavior:

python2 ~/PiSDR/CygnusRFI/CygnusRFI.py will fail because gnuradio libraries are for Python3 (specifically due to type declarations).

python3 ~/PiSDR/CygnusRFI/CygnusRFI.py will fail due to Py2 vs Py3 issues starting with raw_input.  changing that  to input will then fail on execfail, which will then fail on ....

luigifcruz, 2021-02-11:
Thanks for spotting this issue. I think the new GNU Radio version 3.8 only works with Python 3.

Perhaps @0xCoto can update the files in question to Python 3?

0xCoto, 2021-02-11:
I'd suggest installing [Virgo](https://github.com/0xCoto/Virgo), as CygnusRFI itself is now rather deprecated. Virgo is compatible with both Python 2.7/3.x and contains a load of features for radio astronomy, including the (now updated) CygnusRFI for rapidly carrying out RFI surveys. 🙂

0xCoto, 2021-02-11:
Actually please *do* report issues with GR versions! I've heavily tested all of Virgo's GR flowcharts with 3.7, but help is definitely wanted in debugging issues in 3.8 and 3.9, so please report any bugs!

luigifcruz, 2021-02-11:
Thanks for the input @0xCoto! I will apply the changes in the next release. Once it's live I will close this issue.

luigifcruz, 2021-12-30:
Python 2 was removed in the upcoming release (v6.0).

closed 2021-12-30 by luigifcruz