luigifcruz/pisdr-image #52: GLRPT and

benb0jangles, 2021-04-13:
Please can we include XWXAPT (NOAA APT) and GLRPT (METEOR M2 LRPT) in the next package? 
I would be happy to contribute, build, package.

benb0jangles, 2021-04-13:
install GLRPT:

install dependency:
sudo apt-get install libturbojpeg-dev

install glrpt:
cd /home/pi/PiSDR/Software
git clone https://github.com/dvdesolve/glrpt.git
cd glrpt
mkdir build
cmake -DCMAKE_INSTALL_PREFIX=/usr ..
make

Brings error:
/home/pi/PiSDR/Software/glrpt/src/sdr/SoapySDR.c:596:9: error: too many arguments to function ‘SoapySDRDevice_setupStream’

Link:
https://github.com/dvdesolve/glrpt/issues/12

benb0jangles, 2021-04-13:
install GLRPT 2.5.0:

install dependency:
sudo apt-get install libturbojpeg-dev

install glrpt:
cd /home/pi/PiSDR/Software
wget https://github.com/dvdesolve/glrpt/archive/refs/tags/v2.5.0.tar.gz
tar -zxvf v2.5.0.tar.gz
cd glrpt-2.5.0
mkdir build
cd build
cmake -DCMAKE_INSTALL_PREFIX=/usr ..
make

brings error:
/home/pi/PiSDR/Software/glrpt-2.5.0/src/sdr/SoapySDR.c:593:9: error: too many arguments to function ‘SoapySDRDevice_setupStream

https://github.com/dvdesolve/glrpt/issues/13

benb0jangles, 2021-04-13:
SoapySDR info:

pi@raspberrypi:~ $ ls /usr/include | grep -i soapy

pi@raspberrypi:~ $ ls /usr/lib | grep -i soapy

pi@raspberrypi:~ $ ls /usr/local/include | grep -i soapy
soapy
SoapySDR

pi@raspberrypi:~ $ ls /usr/local/lib | grep -i soapy
libgnuradio-soapy.so
libgnuradio-soapy.so.2.1.3git
libgnuradio-soapy.so.95d9dd00
libSoapySDR.so
libSoapySDR.so.0.8
libSoapySDR.so.0.8.0
SoapySDR

benb0jangles, 2021-04-13:
install GLRPT 2.5.2:

 cd /home/pi/PiSDR/Software
wget http://www.5b4az.org/pkg/lrpt/glrpt-2.5.2.tar.bz2
tar jxvf glrpt-2.5.2.tar.bz2 
cd glrpt-2.5.2
./autogen.sh 
./configure 
make

error: 
sdr/SoapySDR.c:511:9: error: too many arguments to function ‘SoapySDRDevice_setupStream’
   ret = SoapySDRDevice_setupStream(

benb0jangles, 2021-04-13:
advise to fix SoapySDR

benb0jangles, 2021-04-13:
XWXAPT:
cd /home/pi/PiSDR/Software
wget http://www.qsl.net/5b4az/pkg/apt/xwxapt/xwxapt-3.4.2.tar.bz2
tar jxvf xwxapt-3.4.2.tar.bz2 
cd xwxapt-3.4.2
./autogen.sh
./configure
make
sudo make install -j4

Run:
xwxapt

info:
https://www.qsl.net/5b4az/pkg/apt/xwxapt/xwxapt.html#Compilation

luigifcruz, 2021-04-13:
The problem seems to be related with a version mismatch between the SoapySDR and GLRPT. The image runs on the latest SoapySDR available. The GLRPT needs to be updated in this case.

benb0jangles, 2021-04-13:
> The problem seems to be related with a version mismatch between the SoapySDR and GLRPT. The image runs on the latest SoapySDR available. The GLRPT needs to be updated if this case.

https://github.com/dvdesolve/glrpt/issues/13

luigifcruz, 2021-04-13:
I'll consider adding GLRPT and XWXAPT to the next version.

benb0jangles, 2021-04-15:
There is also MLRPT a command line Meteor M2 capture-demodulate-decoder:

install dependency:
sudo apt-get install libturbojpeg-dev

MLRPT:
cd /home/pi/PiSDR/Software
mkdir mlrpt
cd mlrpt
wget https://github.com/dvdesolve/mlrpt/archive/refs/tags/v1.8.0.tar.gz
tar jxvf v1.8.0.tar.gz
cd v1.8.0
./autogen.sh
./configure
make
sudo make install

Check works with:
mlrpt -help

https://github.com/dvdesolve/mlrpt/tree/v1.8.0

luigifcruz, 2021-12-30:
GLRPT and XWXAPT were included in the upcoming release. Thanks for contributing!

closed 2021-12-30 by luigifcruz