luigifcruz/pisdr-image #54: Issues installing gr-gsm, which relies on gr-osmosdr, which relies on a newer gnuradio version ?

buforded, 2021-04-16:
Hello,

I am using 2020-11-13-PiSDR-vanilla.img.

My goal is to run gr-gsm on my limeSDR-USB device.

I attempted to use:

https://gitlab.com/myriadrf/gr-gsm

... which is a gr-gsm built for limeSDR.  However, this requires gr-osmosdr.  When I attempt to compile gr-osmosdr, I bomb out with errors telling me that it requires GNU Radio 3.9.

Is this a very simple issue of pisdr having a slightly older GNU Radio (3.8) than gr-osmosdr is looking for ?

May I suggest that the following items would be useful to include in the pisdr distribution:

- kal/kalibrate for each of the physical SDR devices (kal is *very* useful)
- gr-gsm
- basic osmocom libraries, etc., for other osmocom utilities and tools that users might install.

Thank you very much.

buforded, 2021-04-23:
I guess the question is, is it safe to update/upgrade gnuradio on pisdr without trashing the system and all of the rest of the dependencies ?  Any guidance would be appreciated.

Thank you.

luigifcruz, 2021-12-30:
The GNURadio installation was updated to version 3.9 in the upcoming release (v6.0). This should solve your problem. Thanks for contributing!

closed 2021-12-30 by luigifcruz