luigifcruz/pisdr-image #58: PlutoSDR not working as expected.

luigifcruz, 2021-05-28:
Command `iio_info -s` throws `unable to create local iio context: no such file`. May be related to a `config.txt` file in Pluto.

```
[NETWORK]
hostname = pluto
ipaddr = 192.168.1.64
ipaddr_host = 192.168.1.85
netmask = 255.255.255.0
[USB_ETHERNET]
ipaddr_eth = 192.168.1.8
netmask_eth = 255.255.255.0
```

luigifcruz, 2021-12-30:
All PlutoSDR packages were updated to the latest version for the upcoming release (v6.0).

closed 2021-12-30 by luigifcruz

NurreNx, 2022-01-08:
Hello, I just tried executing the command `iio_info -s`  on your latest release (6.0) and I still got the error `unable to create local iio context: no such file` and then it show 2 available context, one with usb and another with an IP adress. Maybe the two context conflict with each other ?

luigifcruz, 2022-01-09:
Hi, according to the [Analog Devices forum](https://ez.analog.com/linux-software-drivers/f/q-a/543322/unable-to-create-local-iio-context-no-such-file-or-directory-2-on-ubuntu-20-4), this message can be ignored. I can confirm I can use my PlutoSDR with the image.