luigifcruz/pisdr-image #79: XRDP Not Working

euangordon, 2022-03-30:
I prefer to use RDP rather than VNC to connect from Windows to RPI.

On a Pi 400, with standard Raspian, installing xrdr just works and Windows can connect. But in this image xrdp is not working. I have tried removing VNC Server to see if that helped but still now luck. 

When XRDP tries to run:

[ERROR] xrdp_mcs_send: xrdp_iso_send failed
Mar 30 17:14:29 pisdr xrdp[2317]: [ERROR] xrdp_sec_send: xrdp_mcs_send failed
Mar 30 17:14:29 pisdr xrdp[2317]: [ERROR] xrdp_rdp_send_data: xrdp_sec_se...iled
Mar 30 17:14:29 pisdr xrdp[2317]: [ERROR] Sending [MS-RDPBCGR] TS_SHUTDOW...iled
Mar 30 17:14:29 pisdr xrdp[2317]: [ERROR] xrdp_sec_recv: xrdp_mcs_recv failed
Mar 30 17:14:29 pisdr xrdp[2317]: [ERROR] xrdp_rdp_recv: xrdp_sec_recv failed
Mar 30 17:14:29 pisdr xrdp[2317]: [ERROR] libxrdp_process_data: xrdp_rdp_...iled
Mar 30 17:14:29 pisdr xrdp[2317]: [ERROR] xrdp_process_data_in: xrdp_proc...iled
Mar 30 17:14:29 pisdr xrdp[2317]: [ERROR] xrdp_iso_send: trans_write_copy...iled
Mar 30 17:14:29 pisdr xrdp[2317]: [ERROR] Sending [ITU T.125] DisconnectP...iled

I have googled the issue and found nothing so far. Any pointers to help diagnose would be great.

euangordon, 2022-03-30:
Managed to get it working, this helped:
http://www.intellamech.com/RaspberryPi-projects/rpi_vnc_rdp.html

texadactyl, 2022-04-05:
@euangordon Can you close this issue, given you solved your XRDP problem?

closed 2022-04-05 by euangordon