luigifcruz/pisdr-image #83: RTL-SDR V3 not recognized

Jakuko99, 2022-06-13:
**Describe the bug**
After clean install of PiSDR image on RPi

**To Reproduce**
Steps to reproduce the behavior:
1. Ran volk_profile after first boot
2. Rebooted RPi
3. Ran command rtl_test in terminal and got message: *No supported devices found.*
4. Ran command  SoapySDRUtil --probe="driver=rtlsdr" and got error message no device found
5. When launched SDR++ it couldn't find RTL-SDR either

**Expected behavior**
Output parameters of RTL-SDR dongle and be able to use it in the SDR++ program.

**Screenshots**
[image]

**Device Version**
Image used on Raspberry Pi 4 4 GB RAM with genuine RTL-SDR V3 dongle bought directly from rtl-sdr-blog

nzstevem, 2022-06-17:
Hi, I've just loaded a fresh image on a 3B and everything is fine.

[image]

nzstevem, 2022-06-17:
[image]

Jakuko99, 2022-06-17:
Thanks, later I found out it was because of crappy USB extension cable, so this is meaningless now

closed 2022-06-17 by Jakuko99