luigifcruz/pisdr-image #84: GnuRadio - qt.qpa.xcb: QXcbConnection: XCB error: 148 (Unknown),

nzstevem, 2022-06-18:
Downloaded, imaged, and booted on a Raspi 3b the latest 6.1 PiSDR image. Connected to the image with both Putty/ssh and VNC. All appeared good. 

Ran the volk script. All good.

Ran a standard update "sudo get-apt update/upgrade". Some libraries updated.

Ran GnuRadio in design mode via VNC and created very simple model.  All good so far.

However, when executing the model it fails with the following QT error.

"qt.qpa.xcb: QXcbConnection: XCB error: 148 (Unknown), sequence: 191, resource id: 0, major code: 140 (Unknown), minor code: 20."

Executing the generated python on the command line - "python3 test.grc" - produces the same result.

Looking around at various posts on other forums it appears others have had the same issue with QT.  However there is no consensus as to how the issue can be resolved.  

[image]

Has anyone else seen this issue, or can anyone confirm GNUradio models do indeed run fine in their RasPi environment?

Any help appreciated.

Steve

luigifcruz, 2022-06-21:
Hi, thanks for submitting this issue. Looks like this is a error introduced on a new version of Qt. According to [this thread](https://forum.qt.io/topic/136006/qt-qpa-xcb-qxcbconnection-xcb-error-on-raspberry-pi-3b/18), some models of Raspberry Pi are affected. I would recommend waiting for a patch from Qt.

nzstevem, 2022-06-21:
Thanks, yes I saw that. I found another which said complete rebuild from source cleared the issue - a 60 hr effort on the model 3. I’ll see if I can find it again.

Rgds Steve
021-956-996