luigifcruz/pisdr-image #86: Cheap Chinese RSP1 Msi2500 board only works for seconds on GQRX and SDR++

swissdigitalnet, 2022-10-12:
**Describe the bug**
RSP1 Msi2500 board only works for seconds on GQRX and SDR++

**To Reproduce**
I tried with RTL-SDR dongle V2 and RSP2pro with GQRX and they work ok.
Then I replace the SDR with an RSP1 Msi2500 board. I can select it and the waterfall starts. But after a few seconds the waterfall and the sound stop.

Here is the output of the terminal:
gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.9.5.0-rc1
built-in source types: file rtl rtl_tcp uhd hackrf rfspace airspy airspyhf soapy redpitaya 
gr::log :WARN: file_source0 - file size is not a multiple of item size
Resampling audio 96000 -> 48000
Resampling audio 96000 -> 48000
BandPlanFile is /home/pi/.config/gqrx/bandplan.csv
BookmarksFile is /home/pi/.config/gqrx/bookmarks.csv
[INFO] [UHD] linux; GNU C++ version 10.2.1 20210110; Boost_107400; UHD_3.15.0.0-0-e35f66e8
gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.9.5.0-rc1
built-in source types: file rtl rtl_tcp uhd hackrf rfspace airspy airspyhf soapy redpitaya 
[INFO] devIdx: 0
[INFO] SerNo: B0001P0003
[INFO] hwVer: 1
[INFO] rspDuoMode: 0
[INFO] tuner: 1
[INFO] rspDuoSampleFreq: 0.000000
[INFO] Using format CF32.
Resampling audio 96000 -> 48000
[WARNING] Not updating IFGR gain because AGC is enabled
[WARNING] Not updating IFGR gain because AGC is enabled
gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.9.5.0-rc1
built-in source types: file rtl rtl_tcp uhd hackrf rfspace airspy airspyhf soapy redpitaya 
[INFO] devIdx: 0
[INFO] SerNo: B0001P0003
[INFO] hwVer: 1
[INFO] rspDuoMode: 0
[INFO] tuner: 1
[INFO] rspDuoSampleFreq: 0.000000
[INFO] Using format CF32.
Resampling audio 96000 -> 48000
[WARNING] Not updating IFGR gain because AGC is enabled
[WARNING] Not updating IFGR gain because AGC is enabled

**Expected behavior**
I would like to use it for a training class
**Screenshots**
[image]

**Device Version**
Raspberry Pi 4

**Additional context**
Add any other context about the problem here.

SensorsIot, 2022-10-12:
I used the wrong account. Sorry. This is me. HB9BLA

luigifcruz, 2023-06-25:
Hi,

Apologies for the extremely long delay! The Pi shortage impacted my ability to pump out new releases.

This problem is consistent with a driver failure. I think this could be resolved with a version upgrade.

I plan to release a new PiSDR version soon. I’ll return to this issue once this happens.

Best Regards,
Luigi Cruz

kelogs, 2024-12-04:
@luigifcruz 

Hi,

you do not happen to recall what the issue has been, do you ?

SensorsIot, 2024-12-04:
Hi Luigi,
I am an old man ;-) and do no more remember the details? You can close the issue if you want. However, these boards are quite interesting and very cheap.