luigifcruz/pisdr-image #88: Welcome to PiSDR Discussions!

andythegrouch, 2023-01-28:
### Discussed in https://github.com/luigifcruz/pisdr-image/discussions/42

<div type='discussions-op-text'>

<sup>Originally posted by **luigifcruz** December  8, 2020</sup>
## 👋 Welcome!
  We’re using Discussions as a place to connect with other members of our community. We hope that you:
  * Ask questions you’re wondering about.
  * Share ideas.
  * Engage with other community members.
  * Welcome others and are open-minded. Remember that this is a community we
  build together 💪.

  To get started, comment below with an introduction of yourself and tell us about what you do with this community.
</div>

andythegrouch, 2023-01-28:
Has anybody successfully used a Pluto with this build and if so what RF bandwidth could they achieve. I want to use a Raspi #4 and a Pluto for a project in GRC, but appreciate the Pluto is USB 2 and limits this. But I don’t really want to pay the price of an Ettus B200 to get the 10MHz i need.

luigifcruz, 2023-01-29:
USB 2.0 should be more than enough to stream data at 10 MHz. But the CPU power of the RPi4 might be a limiting factor. I've used Pluto with an RPi4 before but I can't remember a number. Keep in mind that this will be heavily influenced by the amount of DSP you need.

andythegrouch, 2023-01-29:
Hi Luigi,

Thanks for the help. 

Best Regards

Andy

andythegrouch, 2023-01-29:
Hi Luigi,
I have got the Pi up and running, but there is no IIO block in GRC, just a Soapy Pluto sink and source. Is that correct. IIO scope works though so not sure if it is user error.

Andy

luigifcruz, 2023-01-30:
Hey! The [gr-iio](https://github.com/analogdevicesinc/gr-iio) module is not installed automatically in the current version. But it can be installed manually quite easily. I will make sure to add this to the next version.

andythegrouch, 2023-02-01:
Hi Luigi,

I will try to install it using the AD instructions and let you know if there any issues. I think with its price point and performance, once it has been hacked to extend its frequency range and to have 2xTx and 2xRx with the Rev C,the Pluto is probably the best amateur choice at the moment, so you might see a surge of people downloading your image. As an RF and Electronics expert, but not a SW expert,  some of the installations of APIs can be a bit daunting. The reason I have jumped onto my pi 400 is that I have I have had issues with PC and OSX installations and really want to use GRC to try stuff on the Pluto and then add extra functionality in the Python it outputs.

Andy

Jplify, 2023-07-25:
when I try to open any of the software it opens up geany instead.