luigifcruz/pisdr-image #90: Enhancement request: Need a way to update from github w/o reimaging

texadactyl, 2023-02-01:
Or is this solved already and I missed it?  (-:  This could be an interesting project in itself.

luigifcruz, 2023-02-01:
That is a pretty deep rabbit hole. 😅 

It's technically possible to re-run the Ansible scripts on an installed image with some modifications. But sometimes this means that you need to compile GNU Radio from source (or other large programs), which can be a very interesting experience on the Pi.

Another more sustainable alternative is to ship `.deb` files. But this would probably be problematic because they would collide with official packages from maintainers.

It's a tricky situation to navigate.

texadactyl, 2023-02-01:
"Pi" in the sky, eh?  (: