luigifcruz/pisdr-image #92: Upgrade GnuRadio to 3.10.2.0

andrewgreenberg, 2023-02-09:
pisdr ships with GnuRadio 3.9, but GnuRadio has been at version 3.10.2.0 since April 2022. 3.10 has a lot of really great core upgrades, it would be great to have a new pisdr build with the current stable version! Thanks!

texadactyl, 2023-02-09:
It looks like only one file is affected:
```pisdr-image/stage4/13-pisdr-gnuradio/01-run.sh```

```--branch maint-3.9``` --> ```--branch maint-3.10```

aSmig, 2023-02-10:
@texadactyl I'm confused by the steps suggested in your last post. It looks like they pull the latest GNURadio source but don't compile it. I would expect the volk_profile at the end to use the system installed 3.9 libraries. Indeed, running gnuradio-companion loads up 3.9.5.0-rc1 after these steps. Am I missing something?

texadactyl, 2023-02-10:
@aSmig Thanks, you are spot on. Doing too many things at the same time. The Python scripts are updated but the cmake steps need to be completed.  Sorry for the distraction.

aSmig, 2023-02-12:
Based on earlier notes from @texadactyl and various tips found elsewhere, I have 3.10 running on a PiSDR. The most important steps are below. However, it was a journey so these may not be complete instructions. Creating a swap file cuts compile time almost in half on my 8GB Pi 4. Start by logging into your Pi.

```bash
sudo -i
apt update
apt upgrade
apt install libclang-cpp11 liborc-0.4-dev-bin libxapian30 libspdlog-dev
dpkg -r python3-pygccxml
pip install pygccxml

fallocate -l 2G /swapfile
chmod 600 /swapfile
mkswap /swapfile
swapon /swapfile

cd PiSDR/Software/gnuradio
git remote set-branches --add origin maint-3.10
git fetch origin maint-3.10
git switch maint-3.10
git submodule update --init --recursive
mkdir -p build
cd build
cmake -GNinja -DCMAKE_BUILD_TYPE=Release ..
time ninja install
ldconfig
exit

volk_profile
```

cmake takes about 2 minutes
ninja takes about 2 hours 15 minutes
volk_profile takes about 5 minutes

luigifcruz, 2023-06-25:
Hi,

Sorry for the delay. The Pi shortage has affected my ability to release new versions. I plan to release a new version with the latest software soon.

I believe it’s very difficult to compile GNURadio locally with the Pi. Even if successful, the out of tree modules and SDR drivers might also break. Therefore, a full district upgrade is necessary.

Best Regards,
Luigi Cruz