luigifcruz/pisdr-image #95: I am having issues with the rasberrypi accepting this as a operating system. My rapberry pi4 shows no connection to firmware. I used the rapberry pi imager. Is there something else I possibly need to do to get this to load?

KC9RFV, 2023-02-19:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Device Version**
Which Raspberry Pi version this problem happens.

**Additional context**
Add any other context about the problem here.

KC9RFV, 2023-02-19:
I finally got it working by doing nothing different than what it said to do. Not sure what I was doing wrong earlier, but now have it running.

closed 2023-02-19 by KC9RFV

texadactyl, 2023-02-19:
@KC9RFV 
Please close this issue.