luigifcruz/pisdr-image #98: No boot on PiSDR 6.1 RPi 4 2GB

a-oxide, 2023-04-28:
**Describe the bug**
Rpi 4 2gb will not boot after flashed. Device has a keyboard and the official RPi 7" touchscreen attached. 

**To Reproduce**
Steps to reproduce the behavior:
1. Flash SD Card
2. Insert into Pi
3. Attempt to boot
4. See black screen, or occasional blinking "_"

**Expected behavior**
The Pi to boot into desktop

**Device Version**
RPi 4 2GB

**Additional context**
I have tried downloading multiple versions, redownloading version 6.1, booting from a battery and from an official RPi power supply. I have tried flashing with both BalenaEtcher and RPI Imager. The screen consistently boots into a black screen after the initial rainbow box.

luigifcruz, 2023-05-21:
Hi, thanks for trying PiSDR. Is the behavior the same as the original Raspberry Pi OS? I don't have a screen so I can’t validate it works with PiSDR.

a-oxide, 2023-05-21:
I’m afraid not. Thank you for getting back to me!

luigifcruz, 2023-05-21:
No problem! Interesting. I can't remember any setting to turn this on by default in the image builder. Perhaps there might be a setting that you can enable in `rasp-config`.

a-oxide, 2023-05-21:
Hmm. I’ll dig around some more. Thanks!

JoeMatt, 2023-06-28:
Same problem on 3B+. It starts to load desktop and then blacks out. Using the older 7" touchscreen for the 3B+, latest release with Balana.

The first boot it went to resizing and rebooted after that was complete. So the screen was working then.
Llater in the second boot it gets stuck. You can nearly catch the line if you turn off the boot screen.

Not sure if it's X or init.d process but it hard locks aftwards.

I'll update with a video once it's rendered.

JoeMatt, 2023-06-28:
FYI: I ran raspi-config and turned on all advanced graphics options and GL except wayland. Rebooted and X performance and SDR++ rendering was way better.

Screen does blank out mid boot for a second but comes back up to X as normal. Touch works too.